vjmuzik/NativeEthernet #2: Ethernet.begin() issue

bvanpoortvliet, 2020-07-05:
Hi,

Whenever we try to run a example sketch without the PHY connected to a network, Ethernet.begin(mac) takes approximately 60000 milliseconds to start. If Ethernet.begin(mac, ip) it takes about 18 minutes to start. The moment a network jack is plugged in, Ethernet.begin() immediately executes.

Is there any way to overcome Ethernet.begin() getting stuck while the board is not connected to a network?

Kind regards

shmorgan, 2020-07-19:
I have verified that if there is a Static IP address set with the Std Arduino Wiznet Library, the begin statement will NOT block during startup with no link available.

bvanpoortvliet, 2020-08-01:
Although I'm not entirely sure what you mean, I did find a fix for my issue here:

[PJRC Forum](https://forum.pjrc.com/threads/60857-T4-1-Ethernet-Library/page3)

Uncommenting or removing  ` while(!link_status){}` in NativeEthernet.cpp around line 292 fixed my issue.

closed 2020-08-01 by bvanpoortvliet

mhusinsky, 2020-11-13:
thanks for this hint @bvanpoortvliet !
i'd argue to make this behaviour (blocking or not) configurable from outside.