vjmuzik/NativeEthernet #3: Teensy 4.1 MQTT

janvankeulen, 2020-07-07:
I cannot get this library to work with the Pubsubclient library to connect to MQTT over ethernet. I used the example from pubsubclient and replace Ethernet.h with NativeEthernet.h. Is there anything more I need to do?
Other Ethernet functionality works 😕

dondipietro, 2020-07-18:
Hello,
I tried with : ArduinoMqttClient.h as mqtt client
On base of the exemple "webclient" with NativeEthernet.h.
And same problem it stay on attempting to connect to mqtt broker.

System get an ip

`/*
  Web client

 This sketch connects to a website (http://www.google.com)
 using an Arduino Wiznet Ethernet shield.

 Circuit:
 * Ethernet shield attached to pins 10, 11, 12, 13

 created 18 Dec 2009
 by David A. Mellis
 modified 9 Apr 2012
 by Tom Igoe, based on work by Adrian McEwen

 */

#include <SPI.h>
#include <NativeEthernet.h>
#include <ArduinoMqttClient.h>
#define mqttport 1883
const char broker[]      = "192.168.69.55";

// Enter a MAC address for your controller below.
// Newer Ethernet shields have a MAC address printed on a sticker on the shield
byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };

// if you don't want to use DNS (and reduce your sketch size)
// use the numeric IP instead of the name for the server:
//IPAddress server(74,125,232,128);  // numeric IP for Google (no DNS)
char server[] = "www.google.com";    // name address for Google (using DNS)

// Set the static IP address to use if the DHCP fails to assign
IPAddress ip(192, 168, 0, 177);
IPAddress myDns(192, 168, 0, 1);

// Initialize the Ethernet client library
// with the IP address and port of the server
// that you want to connect to (port 80 is default for HTTP):
EthernetClient client;
MqttClient    mqttClient(client); 

void connectMQTT() {  
  Serial.print("Attempting to MQTT broker: ");  
  Serial.print(broker);  
  Serial.println(" ");  
  //mqttClient.setUsernamePassword(AUTHMETHOD,AUTHTOKEN); //user pass
  while (!mqttClient.connect(broker, mqttport)) {  
    // failed, retry  
    Serial.print(".");  
    delay(5000);  
  }  
  Serial.println();  
  
  Serial.println("You're connected to the MQTT broker");  
  Serial.println();  
  
  // subscribe to a topic  
  //mqttClient.subscribe("arduino/incoming");  
} 

// Variables to measure the speed
unsigned long beginMicros, endMicros;
unsigned long byteCount = 0;
bool printWebData = true;  // set to false for better speed measurement

void setup() {
  // You can use Ethernet.init(pin) to configure the CS pin
  //Ethernet.init(10);  // Most Arduino shields
  //Ethernet.init(5);   // MKR ETH shield
  //Ethernet.init(0);   // Teensy 2.0
  //Ethernet.init(20);  // Teensy++ 2.0
  //Ethernet.init(15);  // ESP8266 with Adafruit Featherwing Ethernet
  //Ethernet.init(33);  // ESP32 with Adafruit Featherwing Ethernet

  // Open serial communications and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // start the Ethernet connection:
  Serial.println("Initialize Ethernet with DHCP:");
  if (Ethernet.begin(mac) == 0) {
    Serial.println("Failed to configure Ethernet using DHCP");
    // Check for Ethernet hardware present
    if (Ethernet.hardwareStatus() == EthernetNoHardware) {
      Serial.println("Ethernet shield was not found.  Sorry, can't run without hardware. :(");
      while (true) {
        delay(1); // do nothing, no point running without Ethernet hardware
      }
    }
    if (Ethernet.linkStatus() == LinkOFF) {
      Serial.println("Ethernet cable is not connected.");
    }
    // try to congifure using IP address instead of DHCP:
    Ethernet.begin(mac, ip, myDns);
  } else {
    Serial.print("  DHCP assigned IP ");
    Serial.println(Ethernet.localIP());
  }
  // give the Ethernet shield a second to initialize:
  delay(1000);
  Serial.print("connecting to ");
  Serial.print(server);
  Serial.println("...");

  // if you get a connection, report back via serial:
  
}

void loop() {

 if (!mqttClient.connected()) {  
        // MQTT client is disconnected, connect  
        connectMQTT();  
        } 
        mqttClient.beginMessage("/out");   //net power
        mqttClient.print("{\n\"test\" ");  
        mqttClient.print("\n}");        
        mqttClient.endMessage();  
    // do nothing forevermore:

      delay(1);
    
  }

`

weigu1, 2021-03-04:
Hi,
I have the same problem
I get all the time "MQTT connection failed, rc=-2"

Here is my code:
```C
#include <NativeEthernet.h> //for teensy 4.1
#include <PubSubClient.h>

byte NETWORK_MAC[] = { 0xDE, 0xAA, 0xBE, 0xEF, 0xF3, 0x08 };
IPAddress NETWORK_IP      (192,168,1,31); //static IP

const char *MQTT_SERVER_IP = "192.168.1.60";
const int  MQTT_PORT = 1883;
const char *MQTT_CLIENT_ID = "heating1234";

EthernetClient eth_Client;
PubSubClient MQTT_Client(eth_Client);

void setup() {
  Serial.begin(115200);
  while (!Serial) ; // wait for Arduino Serial Monitor  
  Ethernet.begin(NETWORK_MAC,NETWORK_IP); 
  Serial.println("\nEthernet connected with IP address: ");
  Serial.println(Ethernet.localIP());    
  MQTT_Client.setServer(MQTT_SERVER_IP,MQTT_PORT);  
  while (!MQTT_Client.connected()) { // Loop until we're reconnected
    if (MQTT_Client.connect(MQTT_CLIENT_ID)) { // Attempt to connect
      Serial.print("MQTT connected");    
    }
    else {
      Serial.println("MQTT connection failed, rc=" + String(MQTT_Client.state()));      
      delay(5000);  // Wait 5 seconds before retrying
    }
  }
}
void loop() {}
```

I used the newest Arduino, Teensyduino and libs.
Thanks for help

si-hb, 2022-11-07:
Anyone have any luck? I have the exact same problem. I ported working code from a T3.2 with a Wiznet, to. T4.1 using NativeEthernet. Worked perfect on the T3.2, but on the T4.2 nothing but rc=-2. My code is similar to both examples above. Any other Ethernet stuff I try works fine, just not Pubsubclient and nativeethernet together.

si-hb, 2022-12-29:
Be sure to include a call to `MQTT_Client.loop()` within `void loop() {}`.

For what it's worth, I'd also recommend switching to [QNEthernet](https://github.com/ssilverman/QNEthernet)

karelv, 2023-01-05:
I had luck with defining the server ip like this:
```cpp
const IPAddress MQTT_SERVER_IP(192, 168, 1, 60);
```
in stead of using the string notation.
(I will try QNEthernet later)