vjmuzik/NativeEthernet #4: host_name not declared in EthernetClient

david3a, 2020-08-16:
Just udpated my PLatformio native ethernet to a fresh zip from here and get

.platformio\packages\framework-arduinoteensy\libraries\NativeEthernet\src\NativeEthernetClient.cpp: In member function 'virtual int EthernetClient::connect(const char*, uint16_t)':
C:\Users\david\.platformio\packages\framework-arduinoteensy\libraries\NativeEthernet\src\NativeEthernetClient.cpp:45:5: error: 'host_name' was not declared in this scope
     host_name = host;
     ^
*** [.pio\build\pre_and_post_hooks\lib37c\NativeEthernet\NativeEthernetClient.cpp.o] Error 1

I assume the host_name is only used for TLS as its declared

#if FNET_CFG_TLS
    fnet_tls_desc_t tls_desc = 0;
    bool _tls_en = false;
    const char* host_name = NULL;
    fnet_uint8_t* ca_certificate_buffer = (fnet_uint8_t*)mbedtls_test_ca_crt; /* Certificate data. */
    fnet_size_t ca_certificate_buffer_size = mbedtls_test_ca_crt_len;              /* Size of the certificate buffer. */
#endif

I assume it can be managed by adding "#if FNET_CFG_TLS" around its use

david3a, 2020-08-16:
I fixed  my copy by moving declaration of "    const char* host_name = NULL;
" outside of #ifdef

vjmuzik, 2020-09-28:
I didn’t realize I wasn’t receiving notifications for issues, this did get fixed in a previous commit.

closed 2020-09-28 by vjmuzik