vjmuzik/NativeEthernet #6: client.write doesn't return the actual number of bytes written

nmz787, 2020-09-23:
The official Arduino documentation says the `write` method will return the number of bytes actually written, as is expected for most socket programming. This library does not do that, and I'm thus unable to properly stream voluminous data. Is there another function I can call subsequent to `write` to determine the number of bytes just written?

nmz787, 2020-09-23:
at least that's how it feels experimentally, and how the code reads:
```
size_t EthernetClient::write(const uint8_t *buf, size_t size)
{
    size_t ret = Ethernet.socketSend(sockindex, buf, size);
	if (sockindex >= Ethernet.socket_num) return 0;
	if (ret) return ret;
	setWriteError();
	return 0;
}
```

```
/**
 * @brief	This function used to send the data in TCP mode
 * @return	1 for success else 0.
 */
uint16_t EthernetClass::socketSend(uint8_t s, const uint8_t * buf, uint16_t len)
{
    fnet_ssize_t ret = fnet_socket_send(socket_ptr[s], buf, len, 0);
    if(ret == -1) return 0;
    return  ret;
}
```

it seems to return only 1 or 0 from the comment

vjmuzik, 2020-09-28:
I didn’t realize I’m not receiving notifications for issues. 

I don’t remember specifically testing this for myself, but from my understanding it should be returning the number of bytes written. The fnet_socket_send function return the number of bytes written or -1 if there was an error, so to keep inline with the original Ethernet.socketSend function it returns 0 if there was an error or the number of bytes written. I don’t believe I changed anything in the client.write function so barring anything wrong with socketSend that should work the same.