vjmuzik/NativeEthernet #8: connect() hangs system when client not available

janvankeulen, 2020-11-06:
In the setup() I want to connect to a MQTT broker, this works fine when the broker is online but fails when the broker is not online. I expected that the connect() function returns a false when it cannot connect but instead the system does not get anything back and freezes...

```
bool connectMQTT()
{
  EthernetClient testcl;
  char sIp[15] = "192.168.2.44";
  IPAddress MQTTbrokerIP = parseIP(sIp);
  if(testcl.connect(MQTTbrokerIP, settingMQTTport)){
    return true;
  else{
    return false;
}
```

vjmuzik, 2020-11-06:
Last I checked there are no issues there, the client code has a default timeout of 10 seconds where it’ll wait for a client to connect before determining that nothing is there. There is already a function to set the timeout from your sketch, client.setConnectionTimeout(uint16_t timeout). Can you verify if it’s actually freezing or not by reducing the timeout to something more manageable. 10 seconds was the default for the w5500 library so I left it as is so that the functionality would line up.

janvankeulen, 2020-11-06:
> Last I checked there are no issues there, the client code has a default timeout of 10 seconds where it’ll wait for a client to connect before determining that nothing is there. There is already a function to set the timeout from your sketch, client.setConnectionTimeout(uint16_t milliseconds). Can you verify if it’s actually freezing or not by reducing the timeout to something more manageable. 10 seconds was the default for the w5500 library so I left it as is so that the functionality would line up.

I checked a small test program. And indeed it works...but only in the Arduino IDE, when I copy it into PlatformIO it hangs as before...
Now I am totally lost :-(
```

#include <Arduino.h>
#include <SPI.h>
#include <NativeEthernet.h>

byte mac[] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xED };
EthernetClient testcl;

bool connectMQTT3()
{
  char sIp[15] = "192.168.2.44";
  IPAddress MQTTbrokerIP(192,168,2,44);
  Serial.println("initMQTT");
  int result = testcl.connect(MQTTbrokerIP, 1883);
  delay(200);
  Serial.print("result: ");
  Serial.println(result);
  if (result == 1)
  {
    Serial.print("IFFED ");
    return true;
  }
  else
  {
    Serial.print("NOT IFFED ");
    return false;
  }
}

void setup() {
  Serial.begin(9600);
}

void loop() {
  Ethernet.begin(mac);
  auto link = Ethernet.linkStatus();
  Serial.print("Link status: ");
  switch (link) {
    case Unknown:
      Serial.println("Unknown");
      break;
    case LinkON:
      Serial.println("ON");
      break;
    case LinkOFF:
      Serial.println("OFF");
      break;
  }
  delay(1000);
  connectMQTT3();
}

```

hfahema1, 2021-05-02:
I was able to get around a similar problem by changing the "while" in line 293 of the NativeEthernet.cpp file to an "if". I assumed my program was hanging up there and when I made the change, my program moved on.