vjmuzik/NativeEthernet #12: Setting a static IP hangs until an Ethernet cable is attached to another port

ssilverman, 2021-02-08:
https://github.com/vjmuzik/NativeEthernet/blob/92540fd198f2b635d3af27b738e192b61309287e/src/NativeEthernet.cpp#L293-L294

This code will hang if I want to set a static IP but don't yet have a cable plugged in. I waited for a while and it looks like the link status is never set to something other than `Unknown` (`0`). Is `link_callback` supposed to be called at all in this case?

ssilverman, 2021-02-08:
Related: https://forum.pjrc.com/threads/60857-T4-1-Ethernet-Library?p=266705&viewfull=1#post266705

ssilverman, 2021-02-08:
My solution is to just comment out that tight `while` loop. Then things work as expected.

ssilverman, 2021-02-22:
Just pinging on this issue because you answered #13 :)

troy-at-skyfront, 2021-06-01:
I am also having this issue. I'll try to comment out the while loop as you suggested.

lorenzofattori, 2021-07-20:
Having the same issue, currently commented the lines in the library, but it would be nice to have a fix on library level so we don't have to do this manually at every library update.

Cheers!

ssilverman, 2022-01-30:
I think that this behaviour shouldn’t be here. I know @vjmuzik you said that the library should block because that’s how you remembered some older Arduino library behaving, however, why would there exist an `Ethernet.linkStatus()` function at all if it would always return “LinkON”?

If the library blocks until “link”, wouldn’t that contradict the mere existence of that function? For example, the API says there’s also “LinkOFF” and “Unknown” return values. How can code ever even check this if the library blocks until link?

Additionally, this blocking behaviour makes it very easy to “freeze” any Teensy using this library just by unplugging the Ethernet cable.