vjmuzik/NativeEthernet #14: EthernetClient::read(buf, size) always seems to return size

ssilverman, 2021-02-22:
This function always seems to return the `size` value, no matter what I pass in. If nothing's available, or something less than `size` is available, it should return that number.

Is this function even intended to be called without first checking "socket-available" or some such, and then using that value for `size`?

vjmuzik, 2021-02-22:
Now I do see those errors and I know why, I didn’t add anything to keep track of what’s actually in the socket buffers. But yes you do have to call client.available() before doing reads, without calling it first the socket buffers won’t have data. Unfortunately I don’t have time to work on the fix for this, but it is something that should be fixed. I’ve slowly been learning to just use FNET by itself since I don’t like the standard Arduino Ethernet style. Not to mention the blocking nature that just wastes resources with how fast the Teensy 4.1 is.

ssilverman, 2021-02-22:
Thanks for having a peek. I think I’ll take your advice and try to learn FNET’s non-blocking API. (See also #11 and #12 for Arduino API issues, which, as soon as I learn FNET, will become moot for me, but I think affect functionality.)

curtwelch, 2024-09-01:
Well, I've been hunting down bugs in our use of ArudinoWebSockets on Teensy4.1 and since it uses NativeEthernet I've been chasing this bug hunt today down to bugs in this code.

AurdinoWebSockets makes the error of not always causing client.available().  They tried but didn't get it right if the WebSocket message is too large to fit into one packet. It never calls client.avilable() for the second part and then all hell breaks loose and corrupted data is returned. I don't know why corrupted data is being returned.  Haven't figure that out yet.

I'll write this next problem as it's own issue, but I'll mention it here.

You have the variable _remaining in NativeEthernetClient.cpp to track how many bytes are available to read.  But when they fail to call accept() and just call read(), your code will go ahead and try the read even though _avaibale is 0.  Then when it manages to read data, you blindly subtract the 485 bytes (say) read, from _available, which allows it to go negative,  And once it's negative, available() stop working since it won't try to re-fill the buffers unless _aviable == 0. So it hangs forever, even when calling available() with available(), endlessly returning -485 bytes.